FYSETC/Marlin-2.0.x-FYSETC #6: Can the Cheetah 32bit Board be used in Ender-2?

probonopd, 2019-07-13:
Can the Cheetah 32bit Board be used in Ender-2?

https://de.aliexpress.com/item/33040326590.html

raffosan, 2019-07-17:
technically yes, I have also made a firmware that shoudl work , but I have to fix few issues with the tmc drivers

probonopd, 2019-07-17:
Do Ender-2 vs. Ender-3 have different pinouts on the headers? What pins need to be changed? Thanks.

raffosan, 2019-07-19:
there should be no problems on the pins, if u change the board is all realative on the new board, the only main big difference is that the ender3 is 24 volts and the ender2 12v.
for tmc drivers 24v are better, but so far seems I can work at 12v

trouch, 2019-07-30:
Using some diff tool with Ender-2 sample configuration and Cheetah one, you should be able to make the adjustments required:
- BED_SIZE
- MAX_POS
- Endstop config
- Feedrates
- Acceleration
- Jerk

That's pretty much it :-)

mellda, 2019-08-03:
The display is not working with ender 2. It lights up but there are no pixels. When I connect the display from ender 3, it is ok.

mellda, 2019-08-03:
No need to use step-down module. The fan voltage jumper is not important.

probonopd, 2019-08-04:
> The display is not working with ender 2. It lights up but there are no pixels. When I connect the display from ender 3, it is ok.

Could it have to do with a different pinout on the LCD header?
I read here that the pinout is slightly different:
https://www.3d-druck-community.de/showthread.php?tid=26752&pid=196061#pid196061

| Mainboard  | Ender-2 | Ender-3 |
| --- | --- | --- |
| Pin 1 | Atmega 3 = Arduino 7 |  Atmega 33 = Arduino 28 |
| Pin 8 | Atmega 1 = Arduno 5 | Atmega 20 = Arduno 17 |

imcauseofdeath, 2019-09-12:
> technically yes, I have also made a firmware that shoudl work , but I have to fix few issues with the tmc drivers

Do you have the firmware posted somewhere? I will be looking at putting this board in my Ender 2 when it arrives and am curious about what is needed to get the display working correctly and good firmware. Otherwise, I can do the firmware (or at least try lol) and I I think the link provided for the display will help. We will see. 

Thanks for any help. :)

probonopd, 2020-07-04:
Can the Cheetah 32bit Board be used in Ender-2 now? How?

probonopd, 2020-08-04:
@github-actions this is not resolved yet.

GerogeFu, 2020-08-04:
As raffosan and troch said , you just need to change the configuration and compile the firmware yourself , it should works.
[image]

probonopd, 2020-08-12:
Thank you very much. If anyone ever tries this on real hardware, please leave a comment. Again, thanks a lot.

closed 2020-09-12 by GerogeFu