FYSETC/Marlin-2.0.x-FYSETC #7: Layer fan only turn on when I click on LCD

Eliotht, 2019-07-17:
I've installed Cheetah on Ender 3 and everything is fine but the layer fan don't turn on automatically. Even if I configure Slicer for turn on fan on layer 2, I have to do a simple CLICK on the rotary of LCD for turn on the layer fan.. How can we solve it?

Thanks

konqiDAM, 2019-07-23:
This needs more attention, myself and several more people are having this exact issue, also turning off the fan or any speed change requires a click

Eliotht, 2019-07-23:
I hope they read us because it's a whore

trouch, 2019-07-30:
I've noticed that issue with main Marlin code. Doesn't seems to be specific to FYSETC branch. I'll do some SWD debug eventually.

trouch, 2019-07-30:
Also fan works properly using latest FYSETC build. No LCD click required.

trouch, 2019-08-03:
As a workaround until proper fix, do not enable SPEAKER in your configuration.
There is a conflict between the fan PWM output and buzzer one.

joewys, 2019-08-05:
on every change of layer there is a lcd click whit the new firmware :/

closed 2020-07-05 by github-actions[bot]