FYSETC/Marlin-2.0.x-FYSETC #9: Build fails during fysetc_STM32F1.py

wees, 2019-07-21:
### Description

Running build in PlatformIO fails 

### Steps to Reproduce

1. git clone latest version of Marlin-2.0.x-FYSETC
2. Open in PlatformIO (I'm using OS X VsCode, but it fails in Atom as well) 
3. Without changing any configuration, click Build.
4. Build fails.

**Expected behavior:** Build is successful

**Actual behavior:** Build Fails

#### Additional Information
The Build is failing in buildroot/share/PlatformIO/scripts/fysetc_STM32F1.py If I comment out: 
`extra_scripts = buildroot/share/PlatformIO/scripts/fysetc_STM32F1.py` in platformio.ini, the build is successful.

It's almost as if the function is outputting the help documentation instead of running correctly.

First part of the console:
[image]

Console Continued:
[image]

GerogeFu, 2019-07-25:
I don't have MAC to test that. I seems to be the permission issue. In windows you can use admin to run vscode.

GerogeFu, 2019-07-27:
Refer to this 
https://forum.fysetc.com/d/67-firmware-compile-error/34

closed 2019-07-27 by GerogeFu