FYSETC/Marlin-2.0.x-FYSETC #10: Error when building the project

mellda, 2019-08-03:
I cannot build the firmware.

### Steps to Reproduce

1. clone the project and open it in VCS or Atom
2. change environment to default_envs = fysetc_STM32F1
3. Build

**Expected behavior:** [What you expect to happen]

**Actual behavior:** There is error 

```
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str':
  File "C:\users\melld\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 134:
    env.SConscript(item, exports="env")
  File "C:\Users\melld\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\melld\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\m\Marlin-2.0.x-FYSETC\buildroot\share\PlatformIO\scripts\fysetc_STM32F1.py", line 32:
    UPLOADER=platform.get_package_dir("tool-stm32duino") + '/stm32flash/stm32flash',
```

trouch, 2019-08-03:
In the platformio.ini, add a platform_package attribute to the fysetc section to install tool-stm32duino package:

```
[env:fysetc_STM32F1]
platform      = ststm32
platform_packages =
  tool-stm32duino
framework     = arduino
board         = genericSTM32F103RC
```

I fixed it in my own branch but haven't pushed it upstream yet.

mellda, 2019-08-03:
Thats amazing! Thank you very much!!! :)

mellda, 2019-08-03:
Now I get this error:

`In file included from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:30:0:
Marlin\src\HAL\HAL_STM32F1\HAL.h:73:4: error: #error "Serial port 0 does not exist"
   #error "Serial port 0 does not exist"
    ^~~~~
In file included from c:\marlin-2.0.x-fysetc\marlin\src\pins\pins.h:69:0,
                 from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
c:\marlin-2.0.x-fysetc\marlin\src\pins\ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~`

mellda, 2019-08-03:
@trouch Please help. I have no idea how to fix it.

eptak, 2019-08-03:
```
#define SERIAL_PORT 1
```

mellda, 2019-08-03:
I copied the `Configuration.h` and `Configuration_adv.h` from `\config\examples\FYSETC\Cheetah\BLTouch\` to `\Marlin\` and then I got this error:

`In file included from c:\m\marlin\src\inc\marlinconfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:
c:\m\marlin\src\inc\SanityCheck.h:384:4: error: #error "POWER_SUPPLY is now obsolete. Please remove it from Configuration.h."
   #error "POWER_SUPPLY is now obsolete. Please remove it from Configuration.h."
    ^~~~~
In file included from c:\m\marlin\src\inc\marlinconfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\HAL_Servo_STM32F1.cpp:26:
c:\m\marlin\src\inc\SanityCheck.h:384:4: error: #error "POWER_SUPPLY is now obsolete. Please remove it from Configuration.h."
   #error "POWER_SUPPLY is now obsolete. Please remove it from Configuration.h."
    ^~~~~`

So I comented out this part:

```
#define POWER_SUPPLY 0

#if POWER_SUPPLY > 0
  // Enable this option to leave the PSU off at startup.
  // Power to steppers and heaters will need to be turned on with M80.
  //#define PS_DEFAULT_OFF

  //#define AUTO_POWER_CONTROL        // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU over this temperature
    #define POWER_TIMEOUT 30
  #endif

#endif
```

and the compilation is ok now. I get these 3 warnings now:

`In file included from Marlin\src\HAL\HAL_STM32F1\HAL_spi_STM32F1.cpp:36:0:
C:\users\melld\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:171:15: warning: '_spi3_this' defined but not used [-Wunused-variable]
 static void (*_spi3_this);
               ^~~~~~~~~~
C:\users\melld\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:168:15: warning: '_spi2_this' defined but not used [-Wunused-variable]
 static void (*_spi2_this);
               ^~~~~~~~~~
C:\users\melld\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:165:15: warning: '_spi1_this' defined but not used [-Wunused-variable]
 static void (*_spi1_this);
               ^~~~~~~~~~`

closed 2019-08-04 by mellda