FYSETC/Marlin-2.0.x-FYSETC #15: Unable to compile the

Jugulaire, 2020-05-28:
### Description

Unable to build the firmware by following the documentation.

### Steps to Reproduce

1. [First Step]
Take linux, like this version for me : 
`jugu@X220:~/Téléchargements/Marlin-2.0.x-FYSETC$ uname -a
Linux X220 4.15.0-91-generic #92-Ubuntu SMP Fri Feb 28 11:09:48 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
`
2. [Second Step]
Open the project in platformIO as explained in the wiki. 
3. [Third step] 
Hit compile, no luck it compile for mega2560 ! 
4.
Then, edit the platformio.ini and switch the default build to STM32F103RC_fysetc. 
5.
Hit build again and have this nice error : 
`
Compiling .pio/build/STM32F103RC_fysetc/src/src/HAL/STM32F1/Servo.cpp.o
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:82:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../pins/ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~
Marlin/src/HAL/STM32F1/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/Servo.cpp:26:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:82:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/STM32F1/Servo.cpp:26:
Marlin/src/HAL/STM32F1/../../inc/../pins/ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~
Marlin/src/HAL/STM32F1/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../HAL/HAL.h:26:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/HAL.h:82:4: error: #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
   #error "SERIAL_PORT cannot be 0. (Port 0 does not exist.) Please update your configuration."
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/STM32F1/../../inc/../pins/ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~
Marlin/src/HAL/STM32F1/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^~~~~
*** [.pio/build/STM32F103RC_fysetc/src/src/HAL/STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RC_fysetc/src/src/HAL/STM32F1/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RC_fysetc/src/src/HAL/STM32F1/HAL_SPI.cpp.o] Error 1
`

GerogeFu, 2020-05-29:
Which board and firmware branch are you using ?  Your board type in configuration.h and default build in platformio.ini should be corresponding. Such as if you define BOARD FYSETC_F6 you need to change default build to mega2560

Jugulaire, 2020-06-01:
Hi, i'm compilling the CHEETAH/V12 Branch. For a Cheetah v1.2B board. 

Now i can compile it but when i upload the Firmware to my board i got an error "TMC CONNEXION ERROR" Looked around on forum from FYSECT but nothing. 

NOTE : I have compiled the firmware with the default platformio.ini file and default configuration.h and configuration_adv.h.

Someone tell to set TMC2208_STANDALONE instead of TMC2208 in the Configuration.h file but it is not working. I got layer shifting which is normal since the TMC2208 need to communicate with the board. 

I've runned this m122 gcode and get error on ALL the TMC2208.

GerogeFu, 2020-06-01:
CHEETAH/V12 is bit old , i will update it soon.
Try CHEETAH/V12-2.0.x branch
https://github.com/FYSETC/Marlin-2.0.x-FYSETC/tree/CHEETAH/V12-2.0.x

Jugulaire, 2020-06-01:
Thx for response, tried with this one. It has failed too. 

Step to reproduce : 
Download the zip from https://github.com/FYSETC/Marlin-2.0.x-FYSETC/tree/CHEETAH/V12-2.0.x

Open it up in platformIO and hit Build. 

I got this error : 

```
Compiling .pio\build\STM32F103RC_fysetc\libb61\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o
.pio\libdeps\STM32F103RC_fysetc\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp: In member function 'bool Adafruit_I2CDevice::read(uint8_t*, size_t, bool)':
.pio\libdeps\STM32F103RC_fysetc\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:160:79: error: no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'
   size_t recv = _wire->requestFrom((uint8_t)_addr, (uint8_t)len, (uint8_t)stop);
                                                                               ^
In file included from C:\Users\jugu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
                 from .pio\libdeps\STM32F103RC_fysetc\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1,
                 from .pio\libdeps\STM32F103RC_fysetc\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:
C:\Users\jugu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: 
note: candidate: uint8 WireBase::requestFrom(uint8, int)
     uint8 requestFrom(uint8, int);
           ^~~~~~~~~~~
C:\Users\jugu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: 
note:   candidate expects 2 arguments, 3 provided
C:\Users\jugu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: 
note: candidate: uint8 WireBase::requestFrom(int, int)
     uint8 requestFrom(int, int);
           ^~~~~~~~~~~
C:\Users\jugu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: 
note:   candidate expects 2 arguments, 3 provided
*** [.pio\build\STM32F103RC_fysetc\libb61\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
```
This error is related to the same library as before : Adafruit BusIO_ID6214

GerogeFu, 2020-06-02:
Comment following line in platformio.ini file
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
And delete folder in Marlin\.pio\build\ PATH
Then compile again.

Jugulaire, 2020-06-02:
Hey, did exactly what you've said. 
But it has failed again. So instead of remove just the `build` file i removed the uild`AND ibdeps`.

Now it compile 
I have some problem to get the flash to work but here is my solution for helping other : 
```bash
# Identify the usr port by it's id 
ls /dev/serial/by-id/
# run the stmflash with the right parameters 
sudo stm32flash -w firmware.bin -v -i rts,dtr /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
```

wester1, 2020-06-04:
Compilation was succesfull but I have 1 yellow remark:
C:\Users\wes\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;

Can I ignore it?

Jugulaire, 2020-06-16:
Yes you can

closed 2020-07-22 by github-actions[bot]